nebulous/infinitesp #4: Any updates on Touch Stat protocols?

Gthubln060920, 2024-05-21:
I have a new Bryant system with Touch stat that is incompaitble with infinitude

I am getting serial data with infinitive, infinitude or finitude but nothing can parse the new messages from my stat.
I thought I saw you mention success with writing to a Touch stat.

Any tips to share?

nebulous, 2024-05-21:
I've been OBE for the last several months, so no progress in automating this - but I was able to get an actual hardware SAM to control my thermostat (which unfortunately requires the stat to go through a full re-installation cycle) as well as changing the current setpoint by writing directly to an extended table. If you're running Infinitude with rs485, you can enable the SCAN_THERMOSTAT option to see the additional tables' data.

jftaylorMn, 2024-05-22:
@Gthubln060920 - I am not familiar with the Bryant touch stats, but they used to be the same as the Carrier.  A couple years ago, Carrier issued firmware that prevented infinitude to work.  That was when they started sending encrypted data (using https) to the carrier mothership in the cloud.  See https://github.com/nebulous/infinitude/issues/148 for a discussion and more information about this,  If being able to intercept the rich data for local use is more important to you than securing the data going to the cloud (I don't use the cloud services so the decision was easy), you might be able to revert to an earlier version of the Bryant firmware and use infinitude.

@nebulous  may have more current information on whether rolling back to a previous firmware would work with more current stats.